SCHUNK-SE-Co-KG/schunk_mechatronic_gripper #55: Driver crashes in transition deactivate or cleanup

jonasbeil, 2025-07-01:
**Problem description**
When using either deactivate or cleanup the driver crashes and restarts.

**Software versions**
 - branch: develop
 - ubuntu22
 - humble

**To Reproduce**
Initial state is active. Try do call on_deactivate() and on_cleanup()

**Expected behavior**
Driver is back in initial configuration

**Additional context**

> [driver.py-1] [DEBUG] [1751374562.784386927] [schunk.driver]: on_activate() is called.
[driver.py-1] [DEBUG] [1751374580.348131841] [schunk.driver]: on_deactivate() is called.
[driver.py-1] Traceback (most recent call last):
[driver.py-1]   File "/opt/ros/overlay_ws/install/schunk_gripper_driver/lib/schunk_gripper_driver/driver.py", line 611, in <module>
[driver.py-1]     main()
[driver.py-1]   File "/opt/ros/overlay_ws/install/schunk_gripper_driver/lib/schunk_gripper_driver/driver.py", line 605, in main
[driver.py-1]     executor.spin()
[driver.py-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 294, in spin
[driver.py-1]     self.spin_once()
[driver.py-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 806, in spin_once
[driver.py-1]     self._spin_once_impl(timeout_sec)
[driver.py-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 803, in _spin_once_impl
[driver.py-1]     future.result()
[driver.py-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 105, in result
[driver.py-1]     raise self.exception()
[driver.py-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 254, in __call__
[driver.py-1]     self._handler.send(None)
[driver.py-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 440, in handler
[driver.py-1]     arg = take_from_wait_list(entity)
[driver.py-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 357, in _take_timer
[driver.py-1]     with tmr.handle:
[driver.py-1] rclpy._rclpy_pybind11.InvalidHandle: cannot use Destroyable because destruction was requested
[driver.py-1] [DEBUG] [1751374580.367020561] [schunk.driver]: on_cleanup() is called.

jonasbeil, 2025-07-02:
Problem remains after fix: 

> [driver.py-1] [DEBUG] [1751457201.105082794] [schunk.driver]: on_deactivate() is called.
[driver.py-1] Traceback (most recent call last):
[driver.py-1]   File "/opt/ros/overlay_ws/install/schunk_gripper_driver/lib/schunk_gripper_driver/driver.py", line 611, in <module>
[driver.py-1]     main()
[driver.py-1]   File "/opt/ros/overlay_ws/install/schunk_gripper_driver/lib/schunk_gripper_driver/driver.py", line 605, in main
[driver.py-1]     executor.spin()
[driver.py-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 294, in spin
[driver.py-1]     self.spin_once()
[driver.py-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 806, in spin_once
[driver.py-1]     self._spin_once_impl(timeout_sec)
[driver.py-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 803, in _spin_once_impl
[driver.py-1]     future.result()
[driver.py-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 105, in result
[driver.py-1]     raise self.exception()
[driver.py-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 254, in __call__
[driver.py-1]     self._handler.send(None)
[driver.py-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 440, in handler
[driver.py-1]     arg = take_from_wait_list(entity)
[driver.py-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 358, in _take_timer
[driver.py-1]     tmr.handle.call_timer()
[driver.py-1] rclpy._rclpy_pybind11.RCLError: failed to call timer: timer is canceled, at ./src/rcl/timer.c:249
[driver.py-1] [DEBUG] [1751457201.136491058] [schunk.driver]: on_cleanup() is called.
[ERROR] [driver.py-1]: process has died [pid 73, exit code 1, cmd '/opt/ros/overlay_ws/install/schunk_gripper_driver/lib/schunk_gripper_driver/driver.py --ros-args -r __node:=driver -r __ns:=/schunk --params-file /root/.ros/launch_params_lfpfg4zq --params-file /root/.ros/launch_params_lh51tizf --params-file /root/.ros/launch_params_1t_6m5kb --params-file /root/.ros/launch_params_ugsvzryk'].
[INFO] [driver.py-1]: process started with pid [305]

The issue has to be reopened

stefanscherzinger, 2025-07-03:
@jonasbeil , please send the commit hash on which you are working: `git rev-parse HEAD` and paste it here.

jonasbeil, 2025-07-03:
Commit hash is: d2aaf52f9f014fb24bebf690915c5a7da3497db6

stefanscherzinger, 2025-07-09:
Should be fixed through #63