SCHUNK-SE-Co-KG/schunk_mechatronic_gripper #120: Faster response time for Grip feedback in ROS 2 driver

Jiawei2019, 2025-09-25:
Hi,

May I ask if there’s any chance to optimize the response speed of the ROS 2 driver for the Grip function?
With the current provided version, after the gripper picks up an object, it usually takes about 1–1.5 seconds before returning whether the object has been successfully grasped.

Is there a way to improve this response latency?

Thanks!

stefanscherzinger, 2025-09-26:
Thanks @Jiawei2019 for your interest in the driver!
Yes, the response speed for gripping is a known issue. We'll address this beginning of next week. I'll let you know once it's on the `develop` branch.

stefanscherzinger, 2025-10-21:
Thanks :)
Please comment if still not working.

closed 2025-10-21 by stefanscherzinger

Jiawei2019, 2025-10-21:
Thanks, this fix helped :)